meta-control/mc_mros_reasoner #3: Complete installation and execution instructions in README

chcorbato, 2019-08-21:
We need to be able to deploy the mros1_reasoner in any machine.

As info, this is the excerpt of what @gavanderhoorn did in my machine to get it running
```
  940  mkdir $HOME/ros_kinetic_py3
  941  cd /home/chcorbato/ros_kinetic_py3/
  942  apt install python3-venv libpython3-dev python-catkin-tools
  943  sudo apt install python3-venv libpython3-dev python-catkin-tools
  944  sudo apt clean
  945  sudo apt update
  946  sudo apt-key del 421C365BD9FF1F717815A3895523BAEEB01FA116
  947  sudo -E apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
  948  sudo apt clean && sudo apt update
  949  python3.5 -m venv venv3.5_ros
  950  source venv3.5_ros/bin/activate
  951  pip3 install wheel
  952  pip3 install -U rosdep rosinstall_generator wstool rosinstall
  953  rosinstall_generator --deps --tar --rosdistro=kinetic ros_base > 
  954  rosinstall_generator --deps --tar --rosdistro=kinetic ros_base > ros_base.rosinstall
  955  wstool init src -j8 ros_base.rosinstall
  956  catkin build -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_VERSION=3.5
  957  cd
  958  cat .bashrc 
  959  vim .bashrc 
  960  nano .bashrc 
  961  source .bash
  962  ls -al
  963  cd ..
  964  cd ros_kinetic_py3/
  965  cd mros_ws/src/
  966  git clone git@github.com:rosin-project/cheops.git
  967  cd mros_ws/src/mros1_reasoner/mros1_reasoner/scripts/
  968  python owl_reasoner.py 
  969  mmons-lang3-3.3.2.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/xml-apis-1.4.01.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/sesame-rio-datatypes-2.7.16.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/commons-csv-1.0.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/guice-multibindings-4.0.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/pellet-query-2.4.0-SNAPSHOT.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/pellet-jena-2.4.0-SNAPSHOT.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/pellet-cli-2.4.0-SNAPSHOT.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/commons-codec-1.6.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/pellet-explanation-2.4.0-SNAPSHOT.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/xz-1.5.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/sesame-rio-ntriples-2.7.16.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/stringtemplate-3.2.1.jar:/home/chcorbato/.local/share/virtualenvs/owlready2/lib/python3.5/site-packages/owlready2/pellet/jena-core-2.13.0.jar pellet.Pellet realize --infer-prop-values --infer-data-prop-values --ignore-imports /tmp/tmpuf3wjjzc
  970  Traceback (most recent call last):
  971  python owl_reasoner.py 
  972  Bindings Statuses:
  973  python owl_reasoner.py 
  974  git merge owlready2 
  975  git push
  976  python mros1_reasoner.py 
  977  python owl_reasoner.py 
  978  git fetch
  979  git checkout master
  980  git pull master
  981  git pull
  982  git checkout reasoner_integration 
  983  git rebase master
  984  git rebase --continue
  985  python mros1_reasoner.py 
  986  cd ros_kinetic_py3/
  987  ls -al
  988  catkin clean -y
  989  clear
  990  ls -al
  991  clear
  992  source venv3.5_ros/bin/activate
  993  env | less
  994  env | grep -i ros  less
  995  env | grep -i ros | less
  996  catkin build -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_VERSION=3.5
  997  pip3 install empy
  998  catkin build -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_VERSION=3.5
  999  source devel/setup.bash 
 1000  python
 1001  roscore
 1002  pip3 install defusedxml netifaces
 1003  roscore
 1004  pip3 install owlready2
 1005  cd ..
 1006  mkdir -p ros_py3_ws/src
 1007  rm -rf ros_py3_ws/
 1008  ls -al
 1009  cd mros_ws/
 1010  ls -al
 1011  catkin clean
 1012  catkin clean -y
 1013  ls -al
 1014  catkin config
 1015  catkin config --extend $HOME/ros_kinetic_py3
 1016  catkin config --extend $HOME/ros_kinetic_py3/devel
 1017  clear
 1018  catkin config
 1019  ls -al
 1020  catkin ls
 1021  ls -al
 1022  ls -al src/
 1023  touch src/rosin_diagnosis_experiments/CATKIN_IGNORE
 1024  catkin ls
 1025  catkin 
 1026  catkin b
 1027  catkin clean -y
 1028  catkin config --help
 1029  catkin config --no-extend
 1030  clear
 1031  ls -al
 1032  clear
 1033  ls -al
 1034  cd ..
 1035  mkdir -p abb_mros/src
 1036  cd abb_mros/
 1037  catkin init
 1038  catkin config --extend $HOME/ros_kinetic_py3/devel
 1039  catkin b
 1040  mc
 1041  catkin b
 1042  cd src/
 1043  ls -al
 1044  cd mros1_reasoner/
 1045  ls -al
 1046  git status
 1047  gitk --all
 1048  ls -al
 1049  cd mros1_reasoner/
 1050  ls -al
 1051  nano -w Cm
 1052  nano -w CMakeLists.txt 
 1053  nano -w package.xml 
 1054  cler
 1055  clear
 1056  ls -al
 1057  cd ..
 1058  ls -al
 1059  cd ..
 1060  cd cheops/
 1061  ls -al
 1062  git status
 1063  git branch -av
 1064  gitk --all
 1065  git branch -av
 1066  git checkout graph_manipulation 
 1067  gitk --all
 1068  git checkout master 
 1069  git ft
 1070  git fetch 
 1071  git status
 1072  git checkout graph_manipulation 
 1073  git reset --hard origin/graph_manipulation 
 1074  clear
 1075  gitk
 1076  cd ..
 1077  ls -al
 1078  clear
 1079  catkin clean -y
 1080  catkin b
 1081  source devel/setup.bash 
 1082  python
 1083  clear
 1084  cd 
 1085  source venv3.5_ros/bin/activate
 1086  ls
 1087  cd ros_kinetic_py3/
 1088  ls
 1089  source venv3.5_ros/bin/activate
 1090  cd ../abb_mros/
 1091  source devel/setup.bash 
 1092  roslaunch mros1_reasoner run.launch 
 1093  cd src/mros1_reasoner/
 1094  python owl_reasoner.py
 1095  cd mros1_reasoner/scripts/
 1096  python owl_reasoner.py
 1097  roslaunch mros1_reasoner run.launch 
 1098  python owl_reasoner.py
 1099  roslaunch mros1_reasoner run.launch 
 1100  ifconfig 
 1101  source /opt/ros/kinetic/setup.bash 
 1102  rosrun rqt_graph rqt_graph 
 1103  env | grep -i ros | sort
 1104  rosnode list
 1105  roswtf
 1106  clear
 1107  env | grep -i ros | sort
 1108  export ROS_MASTER_URI=http://192.168.131.1:11311

```

gavanderhoorn, 2019-08-21:
This seems to partly be a duplicate of https://github.com/rosin-project/cheops/issues/5.

Which was already completed.

chcorbato, 2019-08-21:
Thanks Gijs, sorry I was not aware

closed 2019-08-21 by chcorbato

gavanderhoorn, 2019-08-21:
No problem.

It only documents the way Kinetic is built with Python 3 though.

But the rest is basic workspace setup and building.

Might want to write a few words about that.