meta-control/mc_mros_reasoner #6: Problem migrating to Melodic

chcorbato, 2020-04-17:
I tried to emulate Gij's instructions for [Kinetic with Python3 support](https://answers.ros.org/question/237613/how-to-define-ros-kinetic-to-use-python3-instead-of-python27/?answer=331009#post-id-331009), to setup a ROS Melodic installation with Python3 support for the metacontrol reasoner, see branch [melodic](https://github.com/rosin-project/mros1_reasoner/tree/melodic)

I think I managed to properly setup my Python3 virtual environment, build ROS Melodic bare-bones in a workspace, and then overlay my workspace for mros1_reasoner.
However, when trying to launch the reasoner I get:
```
```

gavanderhoorn, 2020-04-17:
Is there anything in the reasoner nodes that needs functionality only supported in Melodic?

If not, it would seem unnecessary to try to build Melodic.

---

I would perhaps just try to use [rospypi/simple](https://github.com/rospypi/simple).

Create a `venv`, just as you've done earlier, run the two install commands shown in the readme -- make sure to install additional packages if/when needed.

Now install whatever else you need for the reasoner.

`roslaunch` is probably not needed. The `.launch` file on starts a single node, which you could start as a regular program inside a terminal with the `venv` active.

chcorbato, 2020-04-17:
Solved, I had not sourced the ws for ROS Melodic with Python3 before building the mros1_reasoner ws, and therefore it was not properly setup.
I added this to the instructions.

closed 2020-04-17 by chcorbato

chcorbato, 2020-04-30:
> Is there anything in the reasoner nodes that needs functionality only supported in Melodic?
>
> If not, it would seem unnecessary to try to build Melodic.
>
> I would perhaps just try to use [...]

Thanks for the idea! I opened a new issue for it #13