meta-control/mc_mros_reasoner #10: Estimate reasonable values for the QAs of each FD/configuration

chcorbato, 2020-04-29:


darkobozhinoski, 2020-05-04:
My suggestion here is the define a configuration as a pair of QA= (safetyQA, energyQA)

  | Low | High
-- | -- | --
QA Estimated Safety | [0.0, 0.8) | [0.8, 1.0]
QA Estimatd Energy | [0.0, 1.9) | [1.9, 3.0]
  |   |  
  |   |  
  |   |  
  |   |  
  |   |  
  |   |  
  | Low Power | High Power
Low Safety | s ∈ [0.0, 0.8), e ∈ [0.0, 1.9) | s ∈ [0.0, 0.8), e ∈ [1.9, 3.0)
High Safety | s ∈ [0.8, 1.0), e ∈ [0.0, 1.9) | s ∈ [0.8, 1.0), e ∈ [1.9, 3.0)

We can add a standard values in between the high and low. For each new QA, we will need to enlarge the number of configurations.

darkobozhinoski, 2020-05-14:
On the following [link](https://docs.google.com/spreadsheets/d/12aB35Jj4qmarIq3OWHocATmnDfUa6ftp_j09W5twy1g/edit#gid=2104572098) we present the connection between a configuration (FD) and a Quality attribute.

chcorbato, 2020-05-18:
@darkobozhinoski @marioney 

For each `FD/configuration` I need (to be estimated form https://github.com/rosin-project/metacontrol_sim/issues/26):
- [ ] `name`
- [ ] `QA_safety_expected_value`
- [ ] `QA_energy_expected_value`
- [ ] `QA_expected_utility`

marioney, 2020-05-19:
> * `QA_safety_values`: max and min values, maybe avg too -> to post-process into a QA_safety_expected_value
> 
> *

The estimated values are available in [this spreadsheet](https://docs.google.com/spreadsheets/d/12aB35Jj4qmarIq3OWHocATmnDfUa6ftp_j09W5twy1g/edit#gid=1541039955)

From the raw data it should be possible to estimate the `QA_utility`

marioney, 2020-05-22:
The estimated QA values for all 27 configurations are now in this [spreadsheet](https://docs.google.com/spreadsheets/d/12aB35Jj4qmarIq3OWHocATmnDfUa6ftp_j09W5twy1g/edit#gid=1715992839)

chcorbato, 2020-05-22:
@marioney @darkobozhinoski which value do you think we should use as `FD.hasQAestimation` for each configuration?
The `avrg`? The `Max` or `Min`?

darkobozhinoski, 2020-05-22:
I think` Min for safety `and `Max for energy` would be the best.
`Avrg` can also fine.

chcorbato, 2020-05-22:
> I think`Min for safety`and `Max for energy` would be the best.

Let's go with this. I close this issue and create one to add add this values to our rossystem models  https://github.com/rosin-project/rosin-experiments/issues/12

closed 2020-05-22 by chcorbato