meta-control/mc_mros_reasoner #11: Decide how to execute configuration

chcorbato, 2020-04-29:
Can be based

chcorbato, 2020-04-29:
@marioney can you copy here the links to the rainbow code for this?

marioney, 2020-04-29:
> @marioney can you copy here the links to the rainbow code for this?

The code used by rainbow is this  

https://github.com/cmu-mars/cmu-robotics/blob/development/cp-gazebo-p2/cp_gazebo/scripts/kill_bringup_amcl.py

chcorbato, 2020-04-29:
Proposal:
- implement component `Rosgraph_manipulator`
- see specification of interfaces in [architecture gdoc](https://docs.google.com/document/d/1BaOxo7_WQ2fUUUWIyw5pE_K_q6sErQJXge5VeFjyhgQ/edit#heading=h.qljntcabvb0p)

It is based on [`cheops_graph_manipulation`](https://github.com/rosin-project/cheops/tree/master/cheops_graph_manipulation) but:
- action goal is a string with the name of the desired configuration
- we need to remove the previous configuration (to decide how)

closed 2020-04-29 by chcorbato