meta-control/mc_mros_reasoner #9: Define the scenario:

chcorbato, 2020-04-29:
- what is the navigation mission
- what is the adaptation trigger?
- what is the adaptation?

chcorbato, 2020-04-29:
- [ ] Define constants for the battery
- [ ] Define A and B

marioney, 2020-04-29:
>     * Define constants for the battery

https://docs.google.com/spreadsheets/d/12aB35Jj4qmarIq3OWHocATmnDfUa6ftp_j09W5twy1g/edit#gid=0

marioney, 2020-04-29:
> * Define A and B

Possible initial Values:  A =  0, 0 (center of the map)
                                     B =   10.5, 5.5  (Exit point)

[image]

marioney, 2020-04-29:
The following values are used to control the battery load cont
```
const_linear_vel = 1.3
const_acceleration = 0.1
const_frequency = 0.04
additional_consumption = 0.0 

```
With these values, the average power load while running the navigation is
```
safe configuration ~= 0.99
standard configuration ~= 1.45
fast configuration ~= 1.99
```
The `additional_consumption` parameter can be used to artificially increase the power load

marioney, 2020-04-29:
The battery constants have been added to the battery power load controller  (see https://github.com/rosin-project/metacontrol_sim/pull/9#issue-410812757)

closed 2020-04-29 by marioney