meta-control/mc_mros_reasoner #13: Release mros_reasoner as a rospy package without ros installation

chcorbato, 2020-04-30:
From @gavanderhoorn (thanks!) in #6 

> I would perhaps just try to use rospypi/simple.
> 
> Create a `venv`, just as you've done earlier, run the two install commands shown in the readme -- make sure to install additional packages if/when needed.
> 
> Now install whatever else you need for the reasoner.
> 
> `roslaunch` is probably not needed. The `.launch` file on starts a single node, which you could start as a regular program inside a terminal with the `venv` active.

This is an interesting idea to simplify the release and deployment of mros_reasoner

gavanderhoorn, 2020-04-30:
I'm not sure I intended for you to *release* `mros_reasoner`. Perhaps my comment wasn't clear.

chcorbato, 2020-04-30:
No, you are clear, I did not chose the right word, release clearly is not.
build?
develop?

chcorbato, 2020-05-01:
@marioney are you available to implement this?

gavanderhoorn, 2020-05-01:
The message dependencies may be a bit tricky to generate btw. If you're current work-around works, you may consider to keep using it.

marioney, 2020-05-01:
> The message dependencies may be a bit tricky to generate btw.

Do you have / know of an example of this. I've just started looking as you say it seems a bit (lot) tricky.

gavanderhoorn, 2020-05-01:
See [the readme](https://github.com/rospypi/simple/tree/28ef7383e1d1bf87f0c5f91c02b6678b6046938f#generate-message-python-package).

chcorbato, 2020-05-01:
Maybe we can postpone this if it is a bit cumbersome, we still have a lot to do for the paper.