meta-control/mc_mros_reasoner #19: Rosinstall error

darkobozhinoski, 2020-05-01:
I am getting the following error:

/home/darko/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/wstool/config_yaml.py:74: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  yamldata = yaml.load(stream)
ERROR in config: Invalid multiproject yaml format in [https://github.com/rosin-project/mros1_reasoner/blob/mvp/mros1_reasoner/mros1_reasoner.rosinstall]: mapping values are not allowed here
  in "<file>", line 115, column 73

Furthermore, the link to rosinstall is also wrong in this command:
wstool init src https://raw.githubusercontent.com/rosin-project/mros1_reasoner/mvp/mros1_reasoner.rosinstall

gavanderhoorn, 2020-05-01:
> Furthermore

actually, that is the reason for the error.

Try this instead:

```
wstool init src -j8 https://github.com/rosin-project/mros1_reasoner/raw/mvp/mros1_reasoner/mros1_reasoner.rosinstall
```

chcorbato, 2020-05-13:
can we close this issue @darkobozhinoski ?

closed 2022-11-18 by Rezenders