meta-control/mc_mros_reasoner #25: Adaptation is not triggered with qa individuals

chcorbato, 2020-05-13:
Objective Status is not updated to ERROR, so the problem is probably in the rules in `tomasys.owl` or in the data types

chcorbato, 2020-05-13:
There is no problem, it is simply that the rule is

**if** `fg.qa.value` **<** `obj.nfr.value`
**then** `obj` in `ERROR`

However, the values coming from diagnostics need to be **normalized**
```
0  - worst case value
1  - desired value
```

chcorbato, 2020-05-13:
@darkobozhinoski are you finally going to normalize safety and energy QA Observations?
I am not sure whether you do it in the `Observer` or I do it in the `mros1_reasoner`

darkobozhinoski, 2020-05-13:
@chcorbato I can do it. Is it ok to normalize according to the following formula:
0.9*normalized_current_speed+0.1_normalized_control_frequency?
Same formula I am using to calculate energy in the configuration space.

chcorbato, 2020-05-13:
That is ok.
For the moment I did not implement top-bottom thresholds semantics in `tomasys`, but simply added a case for the `energy` and `safety` cases in the selection of configurations in mros1_reasoner.py
https://github.com/rosin-project/mros1_reasoner/blob/5263ad6da4a7507754caf6dbf0068e774c223b65/mros1_reasoner/scripts/mros1_reasoner.py#L97-L102

closed 2020-05-13 by chcorbato