meta-control/mc_mros_reasoner #26: Create singularity image for ROS Melodic version with Python3

marioney, 2020-05-14:
Create a singularity image with ROS Melodic version with Python3  to avoid creating a virtual environment.

marioney, 2020-09-01:
> Create a singularity image with ROS Melodic version with Python3 to avoid creating a virtual environment.

This idea is not at good option so we can discard it.

I'm starting to test the [catkin_virtualenv](https://github.com/locusrobotics/catkin_virtualenv) and it seems promising, but it requires further testing.

chcorbato, 2020-09-24:
> I'm starting to test the [catkin_virtualenv](https://github.com/locusrobotics/catkin_virtualenv) and it seems promising, but it requires further testing.

This has been implemented successfully in #85 , so issue solved.

closed 2020-09-24 by chcorbato