meta-control/mc_mros_reasoner #27: Automatic generation of owl file from RosModel

chcorbato, 2020-05-20:


ipa-nhg, 2020-05-20:
Some pointers from me:
Input files: https://github.com/rosin-project/rosin-experiments/tree/master/metacontrol_models_experiment/MoveBaseConfigurations (all the .rossystem files)

If we check a concrete example file (f1_v1_r1.rossystem):

```
RosSystem { Name '**f1_v1_r1**'  RosComponents (
    ComponentInterface { name move_base NameSpace 'move_base'
        RosParameters{
            RosParameter 'controller_frequency' { RefParameter 'move_base.move_base.move_base.controller_frequency' value 15},
            RosParameter 'max_vel_x' { RefParameter 'move_base.move_base.move_base.max_vel_x' value 0.3},
            RosParameter 'inflation_radius' { RefParameter 'move_base.move_base.move_base.inflation_radius' value 0.3}
    }})
    Parameters {
        Parameter { name **qa_safety** type Double value **0.0**},
        Parameter { name **qa_energy** type Double value **0.0**}}
    }
```
 We need to extract only the information marked with '**' and use this information to replace the variables "name_of_RosSystem" , "name_of_RosSystem_param_name" and the 0.5 of Carlos's Skeleton code: https://github.com/rosin-project/mros1_reasoner/blob/rosmodel2owl/mros1_reasoner/scripts/rosmodel2owl.py#L36-L42

If I understood correctly, from the 27 rossystem files we will obtain a single owl file. 

I will meet @ipa-hsd at 13:30 to clarify if his parser code: https://github.com/ipa-hsd/rosgraph_monitor/blob/observers/src/rosgraph_monitor/parser.py can be used for this purpose

chcorbato, 2020-05-20:
> If I understood correctly, from the 27 rossystem files we will obtain a single owl file.

That is right!

ipa-hsd, 2020-05-20:
Had a meeting with @ipa-nhg and she explained me what to do. I will come up with a script. @chcorbato probably we can sync later today for changes.

chcorbato, 2020-05-20:
> Had a meeting with @ipa-nhg and she explained me what to do. I will come up with a script. @chcorbato probably we can sync later today for changes.

Perfect!
@ipa-hsd I'm fully available, just let me know what works better for you to sync (here, hangouts, jitsi, skype...)

chcorbato, 2020-05-22:
The transformation is running in 31def16, so I close this one.

closed 2020-05-22 by chcorbato