meta-control/mc_mros_reasoner #28: desired_configuration needs to be sync between metacontrol nodes

chcorbato, 2020-05-20:


ipa-nhg, 2020-05-20:
to me the best solution will be to set a new parameter to publish (from the metacontrol, right?) the new desired configuration. For example:

`rosparam set /current_configuration "f1_v1_r1" ` 
(obviously from code and not from command line)

>     * `ros_graph_parser` (and [`rosgraph_monitor`](https://github.com/rosin-project/rosgraph_monitor)? ) use it to know what .rossystem to compare with the one extracted from the rosgraph

Here is not needed. We can't modify dynamically the `desired.rossystem` that the `rosgraph_monitor` takes as input. The main purpose of the `rosgraph_monitor` for this models experiment is verify that all the required nodes to run my application are up. The current desired.rossystem looks -> https://github.com/rosin-project/rosgraph_monitor/blob/observers/resources/metacontrol_desired.rossystem

chcorbato, 2020-05-20:
> We can't modify dynamically the `desired.rossystem` that the `rosgraph_monitor` takes as input. The main purpose of the `rosgraph_monitor` for this models experiment is verify that all the required nodes to run my application are up.

For the immediate future, I am very interested in being able to change it dynamically, in order for the metacontrol to manage the nodes on the fly, have a dynamic rosgraph (e.g. kill and launch new nodes as required by the mission goals)

chcorbato, 2020-05-22:
> to me the best solution will be to set a new parameter to publish (from the metacontrol, right?) the new desired configuration. For example:
> 
> `rosparam set /current_configuration "f1_v1_r1" `
> (obviously from code and not from command line)

Agreed! I think we should name that rosparam `desired_configuration`, so as not to confuse it with the one the rosgraph_parser generates
- [ ] @marioney has an issue for metacontrol_experiments
- [ ] @chcorbato needs to do it for `mros1_reasoner`

chcorbato, 2020-05-22:
Since it is under control and there is one issue that will finalize the implementation of this, I close this "design issue"

closed 2020-05-22 by chcorbato