meta-control/mc_mros_reasoner #29: Error executing the reasoner when the observers are loaded

ipa-nhg, 2020-05-20:
I get the following error:

```
bad callback: <function callbackDiagnostics at 0x7f392df50e18>
Traceback (most recent call last):
  File "/home/nhg/rosin-paper/rospy3_melodic/src/ros_comm/rospy/src/rospy/topics.py", line 750, in _invoke_callback
    cb(msg)
  File "/home/nhg/rosin-paper/mros1_reasoner_ws/src/mros1_reasoner/mros1_reasoner/scripts/mros1_reasoner.py", line 220, in callbackDiagnostics
    updateQA(diagnostic_status)
  File "/home/nhg/rosin-paper/mros1_reasoner_ws/src/mros1_reasoner/mros1_reasoner/scripts/mros1_reasoner.py", line 238, in updateQA
    fg = onto.search_one(solvesO=onto.search_one(iri="*" + "o_navigateA")) #TODO read objective from diagnostic_status
  File "/home/nhg/rosin-paper/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/namespace.py", line 380, in search_one
    def search_one(self, **kargs): return self.search(**kargs).first()
  File "/home/nhg/rosin-paper/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/util.py", line 62, in first
    if len(self) != 0: return self[0]
  File "/home/nhg/rosin-paper/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/triplelite.py", line 1470, in __len__
    sql, params = self.sql_request()
  File "/home/nhg/rosin-paper/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/triplelite.py", line 1451, in sql_request
    transits, sql, params = self.sql_components()
  File "/home/nhg/rosin-paper/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/triplelite.py", line 1722, in sql_components
    sql = "SELECT DISTINCT q%s.s FROM %s WHERE %s" % (self.target, ", "   .join(tables), " AND ".join(conditions))
AttributeError: target
```

I am pretty sure I have to load somewhere a model (owl?) that set the target/objective, or maybe set the information manually but I don't know where

## STEPS TO REPRODUCE THE ISSUE:

### Installation:

#### mros1_resoner
I followed: https://github.com/rosin-project/mros1_reasoner/tree/mvp
plus:
```
pip3 install catkin_pkg
pip3 install pyyaml
```
#### metacontrol_sim
I followed: https://github.com/rosin-project/metacontrol_sim/tree/MVP_world

### Execution:
```
roslaunch metacontrol_sim MVP_metacontrol_world.launch
roslaunch mros1_reasoner run.launch
rosservice call /load_observer EnergyQualityObserver
rosservice call /load_observer SafetyQualityObserver
```
The error come periodically and just after call the observers services.

marioney, 2020-05-20:
> I am pretty sure I have to load somewhere a model (owl?) that set the target/objective, or maybe set the information manually but I don't know where

I think that's the error, try  adding the onto argument to the reasoner launcher 

### Execution:
```
roslaunch metacontrol_sim MVP_metacontrol_world.launch
roslaunch mros1_reasoner run.launch onto:=mvp.owl
```
The observers should be activated by default with the `MVP_metacontrol_world.launch` launcher.

ipa-nhg, 2020-05-20:
thanks @marioney , now looks much better :tada: 
I still get an error because of the name of the configurations, but it is related to my version of the rosgraph_manipulator which  needs this changes: https://github.com/rosin-project/mros1_reasoner/issues/27

Could you please update the README of this repo to indicate how to parse the OWL file?? thanks!

closed 2020-05-20 by ipa-nhg

chcorbato, 2020-05-20:
> Could you please update the README of this repo to indicate how to parse the OWL file?? thanks!

That was my bad. I'll put it on my list.