meta-control/mc_mros_reasoner #36: Sometimes no QA energy values are received by mros1_reasoner

chcorbato, 2020-05-21:
### TEST
```
```

### OUTPUT
```
```

chcorbato, 2020-05-21:
> Maybe this is related to the messages published in topic `/diagnostics`
> 
> Could it be that the diagnostic messages from `/ekf_localization` are "jamming" the topic?

A temporal fix could be to disable diagnostics for `/ekf_localization`, but is that possible?
I cannot find how, I do not see any ROS param for that in `/ekf_localization`
Ideas are welcome @marioney @ipa-hsd

chcorbato, 2020-05-21:
> I cannot find how, I do not see any ROS param for that in `/ekf_localization`

Found it! http://docs.ros.org/melodic/api/robot_localization/html/state_estimation_nodes.html?highlight=diagnostics#print-diagnostics
now I just need to find in our code where we launch `/ekf_localization` :smile:

gavanderhoorn, 2020-05-21:
Topics cannot be "jammed" actually.

What could happen is that buffers get full, but then I would expect *all* new messages to not make it into the queue.

Without looking into this in more detail -- and reproducing the setup -- there's not much more I can add I'm afraid.

chcorbato, 2020-05-21:
> What could happen is that buffers get full, but then I would expect _all_ new messages to not make it into the queue (not just those from a specific publisher).

Maybe the messages from `/ekf_localization` are causing the buffer to get full, I guess that also affects `rostopic echo` ?

closed 2023-01-16 by Rezenders