meta-control/mc_mros_reasoner #39: RECONFIGURATION UNKNOWN issue running single exp script WITH metacontrol

chcorbato, 2020-05-22:
INFO] [1590138841.922033]:   >> Finished MAPE-K ** EXECUTION **
== RECONFIGURATION UNKNOWN ==

chcorbato, 2020-05-22:
@marioney is there a way to preserve the terminal running `mros1_reasoner` so as to debug this more easily?

marioney, 2020-05-22:
Sure Just add an sleep command before the `exit` in the ./run_single_sim.sh
```
if [ "$launch_reconfiguration" = true ] ; then
	echo "Launching: mros reasoner"
	gnome-terminal --window --geometry=80x24+10+10 -- bash -c "source $PYTHON3_VENV_PATH/venv3.6_ros/bin/activate;
	source $PYTHON3_VENV_PATH/devel/setup.bash;
	source $REASONER_WS_PATH/devel/setup.bash;
	roslaunch mros1_reasoner run.launch onto:=mvp.owl;
        sleep 20;
	exit"
fi
```

chcorbato, 2020-05-23:
This has been solved now that all info is automatically loaded from the rossystem files, 
- `mros1_reasoner` the autogen ontology file
- `rosgraph_manipulator` the autogen yaml file

closed 2020-05-23 by chcorbato