meta-control/mc_mros_reasoner #42: Metacontrol selects configuration based on QAs AND expected performance

chcorbato, 2020-05-23:
In the MAPE-K PLAN step, the metacontroller should select, amongst the FDs that fullfi the energy and safety NFRs, the one with better performance.

That is, Utility function for the metacontrol (the one used for selection) could be: 
U = QAperformance_expected

we could also consider to include QAenergy_expected and QAsafety_expected

darkobozhinoski, 2020-05-23:
I think the current utility looks good and reasonable.

If you want to include all three we might consider this one:
U = 0.7*QAperformance_expected + 0.2 QASafety_expected + 0.1  QAEnergy_expected
but I do not oppose the one mentioned in your comment.

marioney, 2020-05-23:
> That is, Utility function for the metacontrol (the one used for selection) could be:
> U = QAperformance_expected

I also like this one, as safety and energy are requirements, I think we can leave them out of the utility

chcorbato, 2020-05-23:
Ok, let's leave it only with performance for the moment.

closed 2020-05-23 by chcorbato