meta-control/mc_mros_reasoner #48: Reasoning error

chcorbato, 2020-05-23:
Something did not go well, probably inconsistency in the ontology.

### TEST
```
run_single_sim.sh -i 1 -g 1 -n f1_v3_r1 -r true
```

### OUTPUT
```
...
[ERROR] [1590235376.721149]: Reasoning error:
...
```

chcorbato, 2020-05-23:
> When this error is triggered I'm saving the ontology in `~/.ros/error.owl`, so I will inspect it with Protege to check what happened.

The ontology is coherent and consistent, so I'll need to check what is happening with the reasoning
:worried: 
[image]

chcorbato, 2020-05-23:
It could be due to the `callbackDiagnostics` iasserting new facts during reasoning in `timer_cb`
Maybe we can live with this problem, since the wellformedness of the ontology does not seem affected (see previous comment)

Some additional **OUTPUT** that can be relevant:
```
ERROR] [1590236368.127794]: = RECONFIGURATION SUCCEEDED =
[INFO] [1590236368.131871]: Exited timer_cb for metacontrol reasoning
[INFO] [1590236368.141647]: Entered timer_cb for metacontrol reasoning
[INFO] [1590236368.146012]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
* Owlready2 * Running Pellet...
...
[INFO] [1590236368.368852]: received QA observation
ERROR: FG not found
[INFO] [1590236368.377154]: received QA observation
ERROR: FG not found
[INFO] [1590236369.053949]: received QA observation
ERROR: FG not found
[INFO] [1590236369.060824]: received QA observation
ERROR: FG not found
[INFO] [1590236369.070854]: received QA observation
ERROR: FG not found
[INFO] [1590236369.596342]: received QA observation
ERROR: FG not found
[INFO] [1590236370.137616]: received QA observation
ERROR: FG not found
[INFO] [1590236370.143824]: received QA observation
ERROR: FG not found
[INFO] [1590236370.747393]: received QA observation
ERROR: FG not found
[INFO] [1590236371.426202]: received QA observation
ERROR: FG not found
[INFO] [1590236371.437687]: received QA observation
ERROR: FG not found <--------------------------------------------------------------------------
[ERROR] [1590236371.854689]: Reasoning error: <--------------------------------------
[INFO] [1590236371.868696]:      >> Finished ontological reasoning)
```

chcorbato, 2020-05-23:
The problem is that I'm asserting a new FG with a name that cannot be mapped to incoming QA diagnostics:
```
[INFO] [1590236963.642090]:      >> Finished ontological reasoning)

FGs:
fg_new 	objective:  kb.o_navigateA 	status:  None 	FD:  kb.f1_v1_r3 	QAvalues:  []
```

gavanderhoorn, 2020-05-23:
> It could be due to the `callbackDiagnostics` iasserting new facts during reasoning in `timer_cb`

that would be concurrent write access to shared resource. As `rospy` uses separate threads for timers and subscriptions, either use a (lockless) queue or properly write-lock the shared resource.

marioney, 2020-05-23:
I've added a `lock` for the resource on https://github.com/rosin-project/mros1_reasoner/pull/52/commits/8cba195175fadc5ad56ba822b7373901a94599fe

chcorbato, 2020-05-23:
> I've added a `lock` for the resource on [8cba195](https://github.com/rosin-project/mros1_reasoner/commit/8cba195175fadc5ad56ba822b7373901a94599fe)

It seems to solve the issue for me. I closed it.

closed 2020-05-23 by chcorbato