meta-control/mc_mros_reasoner #49: ERROR: FG not found when received QA observation

chcorbato, 2020-05-23:
### TEST
```
run_single_sim.sh -i 1 -g 1 -n f1_v3_r1 -r true
```
### OUTPUT
```
ERROR] [1590236368.127794]: = RECONFIGURATION SUCCEEDED =
[INFO] [1590236368.131871]: Exited timer_cb for metacontrol reasoning
[INFO] [1590236368.141647]: Entered timer_cb for metacontrol reasoning
[INFO] [1590236368.146012]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
* Owlready2 * Running Pellet...
...
[INFO] [1590236368.368852]: received QA observation
ERROR: FG not found
[INFO] [1590236368.377154]: received QA observation
ERROR: FG not found
[INFO] [1590236369.053949]: received QA observation
ERROR: FG not found
[INFO] [1590236369.060824]: received QA observation
ERROR: FG not found
[INFO] [1590236369.070854]: received QA observation
ERROR: FG not found
[INFO] [1590236369.596342]: received QA observation
ERROR: FG not found
[INFO] [1590236370.137616]: received QA observation
ERROR: FG not found
[INFO] [1590236370.143824]: received QA observation
ERROR: FG not found
[INFO] [1590236370.747393]: received QA observation
ERROR: FG not found
[INFO] [1590236371.426202]: received QA observation
ERROR: FG not found
[INFO] [1590236371.437687]: received QA observation
ERROR: FG not found <--------------------------------------------------------------------------
[ERROR] [1590236371.854689]: Reasoning error: <--------------------------------------
[INFO] [1590236371.868696]:      >> Finished ontological reasoning)
```

marioney, 2020-05-23:
> [INFO] [1590236369.070854]: received QA observation
> ERROR: FG not found

I think the problem is the `grounded_configuration` variable, is set at the beginning but is never updated when there's a reconfiguration.

marioney, 2020-05-23:
That's solved in https://github.com/rosin-project/mros1_reasoner/pull/52

chcorbato, 2020-05-23:
I think I fixed this in 8de2e484650154245ed1eae0efebaa9ee711fe1b, see my comment in the pull request.

closed 2020-05-23 by chcorbato