meta-control/mc_mros_reasoner #50: Remove reasoner output from terminal

chcorbato, 2020-05-23:
The Pellet reasoner called with owlready2 produces and output that is annoying when debugging (see below)
I could not figure out how to remove it from the output. Owlready2 documents do not explain how to handle that
I tried this to temporarily remove it form the `stdout`, but it did not work.

Any ideas are welcome @gavanderhoorn @ipa-hsd @ipa-nhg @marioney 

https://github.com/rosin-project/mros1_reasoner/blob/8de2e484650154245ed1eae0efebaa9ee711fe1b/mros1_reasoner/scripts/mros1_reasoner.py#L230

### OUTPUT
```
* Owlready2 * Running Pellet...
    java -Xmx2000M -cp /home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jena-arq-2.10.0.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/antlr-3.2.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jena-iri-0.9.5.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/pellet-2.3.1.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/httpclient-4.2.3.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/antlr-runtime-3.2.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/owlapi-distribution-3.4.3-bin.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jgrapht-jdk1.5.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/xercesImpl-2.10.0.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/slf4j-log4j12-1.6.4.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/xml-apis-1.4.01.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jcl-over-slf4j-1.6.4.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/slf4j-api-1.6.4.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jena-tdb-0.10.0.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/jena-core-2.10.0.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/commons-codec-1.6.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/aterm-java-1.6.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/httpcore-4.2.2.jar:/home/chcorbato/rospy3_melodic/venv3.6_ros/lib/python3.6/site-packages/owlready2/pellet/log4j-1.2.16.jar pellet.Pellet realize --loader Jena --input-format N-Triples --infer-prop-values --infer-data-prop-values --ignore-imports /tmp/tmp6__87twd
* Owlready2 * Pellet took 1.3000226020812988 seconds
* Owlready * (NB: only changes on entities loaded in Python are shown, other changes are done but not listed)
```

gavanderhoorn, 2020-05-23:
[How can I control logging messages printed by Pellet?](https://github.com/stardog-union/pellet/wiki/FAQ#how-can-i-control-logging-messages-printed-by-pellet)

chcorbato, 2020-05-23:
> 
> Pellet uses JDK logging. The verbosity of this logging can be controlled with a configuration file. The distribution includes a sample `configuration,examples/logging.properties`. The location of the configuration file can be specified by setting the system `propertyjava.util.logging.config.file`. The location should be specified as a (relative or absolute) path. For example, the following command would use a local file located in the current directory:
> ```
> java -Djava.util.logging.config.file=my-logging.properties -jar lib\pellet-cli.jar
> ```
> Note that, in the configuration file, in addition to changing the log level for specific loggers, you must change the handler’s level (you can think of it as two levels of severity checks — one at log record generation, one at consumption). You can find more information about JDK logging [here](http://java.sun.com/j2se/1.4.2/docs/guide/util/logging/overview.html).

It was very promising, but I am not familiar at all with java logging and I could not follow the pointers:

- sample `configuration,examples/logging.properties` -> I could not find it
> You can find more information about JDK logging [here(http://java.sun.com/j2se/1.4.2/docs/guide/util/logging/overview.html)

-> link not working anymore