meta-control/mc_mros_reasoner #53: Should the metacontroller performs only 1 reconfiguration?

marioney, 2020-05-23:
Seems that reconfiguration is launched only once per experiment.

For instnce, I'm running

`./run_single_sim.sh -i 1 -g 1 -n f3_v3_r3 -r true -p 2.0 -e 0.8  -s 0.4`

There's a reconfiguration due to safety at the beginning, but even if the `nfr_energy` is not meet at the end of the experiment, there's no reconfiguration launched

@chcorbato  is this expected?

chcorbato, 2020-05-23:
> There's a reconfiguration due to safety at the beginning, but even if the `nfr_energy` is not meet at the end of the experiment, there's no reconfiguration launched
> 
> @chcorbato is this expected?

No, this is not expected, something is going on, I'll take a look.

marioney, 2020-05-23:
Actually it's switching to `f1_v3_r1` which is strange because that one doesn't meet the safety requirements

chcorbato, 2020-05-23:
I cannot reproduce the case, for me no adaptation was triggered :disappointed: 

[log_Metacontrol_sim_23-21_16_39.zip](https://github.com/rosin-project/mros1_reasoner/files/4672401/log_Metacontrol_sim_23-21_16_39.zip)

chcorbato, 2020-05-23:
> Actually it's switching to `f1_v3_r1` which is strange because that one doesn't meet the safety requirements

fixed in #54

chcorbato, 2020-05-23:
> I cannot reproduce the case, for me no adaptation was triggered 😞
> 
> [log_Metacontrol_sim_23-21_16_39.zip](https://github.com/rosin-project/mros1_reasoner/files/4672401/log_Metacontrol_sim_23-21_16_39.zip)

I keep testing this case because it is weird that for me adaptation is not triggered.
energy threshold is violated sometime, but maybe the metacontroller is skipping those due to its freq (2secs). Although not probable because on the time stamp I see more than 2sec above threshold (at least 3-4 the first time) 😕

chcorbato, 2020-05-23:
> There's a reconfiguration due to safety at the beginning, but even if the `nfr_energy` is not meet at the end of the experiment, there's no reconfiguration launched

I am missing the `safety` reconfiguration

@marioney could this be because in my laptop the experiment runs slower? Maybe the robot do not get enough speed to result in low safety score

chcorbato, 2020-05-23:
> > There's a reconfiguration due to safety at the beginning, but even if the `nfr_energy` is not meet at the end of the experiment, there's no reconfiguration launched
> > @chcorbato is this expected?
> 
> No, this is not expected, something is going on, I'll take a look.

I've been testing this only with the `mros1_reasoner` and mock inputs.
- As you mention, there is no 2nd reconfiguration, even if the `qa_energy` is above `nfr_energy`
- To check what is going on, I've saved the owl file that contains the KB after I force qa_energy observed 0.8 and inspected it with Protege. This is what I found
  - In the KB, the `fg_f1_v2_r1` is updated with qa_value 0.8 (the value I sent) [ ] 
  - The objective is not in ERROR, but it should if the reasoning rules were triggered properly
  - If I run the reasoner in Protege with the ontology file... the objective is updated to `ERROR` status!! 😕

chcorbato, 2020-05-23:
> * Even more mysterious, I run the same metacontrol reasoning, but using as input the file with the KB in the status that was not triggering the 2nd reconfiguration (see [56b22d5](https://github.com/rosin-project/mros1_reasoner/commit/56b22d5f122ccafec3bc7d0f7e339b332561abfa)) and... voila, it triggers the 2nd reconfiguration 🤦 So no clue what is going on. Some refresh needed?

Mystery solved!! not exactly a "refreshing" but close. Reasoner inferences are put in a special ontology, which I guess makes those facts "disappear" for the next run of the reasoner (see [explanation in owlready2 doc](https://pythonhosted.org/Owlready2/reasoning.html#running-the-reasoner))
so fixed in 38d08ac632398d8c835bdef0f330d51455103675

Please check @marioney and close the issue

chcorbato, 2020-05-24:
@marioney when running experiments with metacontrol, check if in some more than 2 adaptations are performed and close this issue if you think it is ok

marioney, 2020-05-24:
>  check if in some more than 2 adaptations are performed and close this issue if you think it is ok

I just confirm that 2 adaptations are executed !!!

closed 2020-05-24 by marioney