meta-control/mc_mros_reasoner #60: List required repos

chcorbato, 2020-08-04:
List of minimally required repos
- check dependency on metacontrol messages (interfaces)
- architectural analysis of metacontrol

chcorbato, 2020-08-05:
I think we should add the required repos in https://github.com/rosin-project/metacontrol_experiments
since that is the base repo to run the experiments for the MODELS paper

chcorbato, 2020-08-05:
A candidate list of repositories can be found in #63

darkobozhinoski, 2020-08-05:
Required Metacontrol Repositories:

[mros1_reasoner ](https://github.com/rosin-project/mros1_reasoner/tree/mvp) - A meta-controller implementation for ROS1
[mc_mdl_tomasys](https://github.com/rosin-project/mc_mdl_tomasys/tree/mvp)  - OWL implementation of the TOMASys metamodel 
[metacontrol_msgs](https://github.com/rosin-project/metacontrol_msgs) - ROS message types for metacontrol
[mc_mdl_abb](https://github.com/rosin-project/mc_mdl_abb/tree/mvp)application models in OWL for the MVP scenario.

Initial tests for the metacontroller can be done using ROS Graph Monitor + QA observers.
[ROS Graph Monitor + QA Observers](https://github.com/rosin-project/rosgraph_monitor/tree/observers)

To deploy the whole setup you will need the following repositories:
metacontrol_nav [Metacontrol_nav ](https://github.com/rosin-project/metacontrol_nav) Repo that contains launcher and configuration files for the navigation stack for the ridgeback robot.
[metacontrol_sim](https://github.com/rosin-project/metacontrol_sim/tree/MVP_world) - Repo that contains gazebo simulation and interface to switch to a different configuration
[metacontrol_move_base_configurations](https://github.com/rosin-project/metacontrol_move_base_configurations) - This repo contains autogenerated configurations.

[OPTIONAL]
[Metacontrol_experiments](https://github.com/rosin-project/metacontrol_experiments) - Repo to configure and run a set of experiments

chcorbato, 2020-08-05:
> [metacontrol_move_base_configurations](https://github.com/rosin-project/metacontrol_move_base_configurations) - This repo contains autogenerated move_base configurations.

Please, indicate what repo is needed to create the configurations for a different system/experiment

chcorbato, 2020-08-05:
> **_Initial tests for the metacontroller can be done using ROS Graph Monitor + QA observers._**
> [ROS Graph Monitor + QA Observers](https://github.com/rosin-project/rosgraph_monitor/tree/observers)

Do you mean that the test can be done **without** _ROS Graph Monitor and QA observers._, or they are required both for the tests and the paper experiments?

darkobozhinoski, 2020-08-05:
@chcorbato , yes, I updated the content. I added the repo that allows generation of new configurations.
I mean that to test the metacontroller you need only the observes and the ros graph monitor. To make it clear, I added this repo in the deployment of the whole setup.

chcorbato, 2020-08-05:
> I mean that to test the metacontroller you need only the observes and the ros graph monitor. To make it clear, I added this repo in the deployment of the whole setup.

You can test it without them, you just need to publish test messages in the topics they do,  e.g. using `rostopic pub`

chcorbato, 2020-09-02:
@darkobozhinoski any update on this issue? Is there anything pending?

chcorbato, 2020-09-24:
This has been solved for installation by @marioney in 67d4af8
But it needs to be properly documented (for maintenance) in the [README.md](https://github.com/tud-cor/mc_mros_reasoner/blob/master/README.md)
Could you please do that @darkobozhinoski ?

closed 2023-01-16 by Rezenders