meta-control/mc_mros_reasoner #65: Objective in Error

darkobozhinoski, 2020-08-05:
ObjectiveError if I run a single run in the metacontrol experiments: 

`./run_single_sim.sh -p 1 -g 2 -n "f1_v1_r1" -o 2 -r "true" -p 1.2`

Same happens if I run only the reasoner: `roslaunch mros1_reasoner run.launch onto:=kb.owl`
[image]

chcorbato, 2020-08-05:
Looks like the metacontroller is not getting or updating the QA values from the observers (if they are published):
``

darkobozhinoski, 2020-08-05:
@chcorbato, do you face the same issue?

chcorbato, 2020-08-05:
Yes, I get exactly the same that you posted here.

marioney, 2020-08-06:
Seems like is working for me. 

The `objetives in error []` message.  Should not be in red, it's just that all objectives are OK.

If I run this

`./run_single_sim.sh -p 1 -g 2 -n "f1_v2_r2" -o 2 -r "true" -p 1.7`

I get a good run, with a successful reconfiguration.

chcorbato, 2020-08-06:
> I get a good run, with a successful reconfiguration.

@marioney can you please add documentation of the expected result? https://github.com/rosin-project/metacontrol_experiments/issues/13

darkobozhinoski, 2020-08-06:
@marioney , can you share a list of active topics. I publish all relevant topics relevant information for energy and safety. However, /obj_error is also active.
This is the list of topics that are published:

```
/amcl_pose
/clicked_point
/clock
/cmd_vel
/d_inflation
/d_obstacle
/diagnostics
/front/scan
/gazebo/link_states
/gazebo/model_states
/gazebo/parameter_descriptions
/gazebo/parameter_updates
/gazebo/set_link_state
/gazebo/set_model_state
/gazebo_ros_control/pid_gains/front_left_wheel/parameter_descriptions
/gazebo_ros_control/pid_gains/front_left_wheel/parameter_updates
/gazebo_ros_control/pid_gains/front_right_wheel/parameter_descriptions
/gazebo_ros_control/pid_gains/front_right_wheel/parameter_updates
/gazebo_ros_control/pid_gains/rear_left_wheel/parameter_descriptions
/gazebo_ros_control/pid_gains/rear_left_wheel/parameter_updates
/gazebo_ros_control/pid_gains/rear_right_wheel/parameter_descriptions
/gazebo_ros_control/pid_gains/rear_right_wheel/parameter_updates
/ground_truth/state
/imu/data
/imu/data/accel/parameter_descriptions
/imu/data/accel/parameter_updates
/imu/data/bias
/imu/data/rate/parameter_descriptions
/imu/data/rate/parameter_updates
/imu/data/yaw/parameter_descriptions
/imu/data/yaw/parameter_updates
/initialpose
/joint_states
/laser_filter/shadows/parameter_descriptions
/laser_filter/shadows/parameter_updates
/map
/map_metadata
/map_updates
/mobile_base/commands/charge_level
/mobile_base/commands/charge_level_mwh
/move_base/NavfnROS/plan
/move_base/TrajectoryPlannerROS/cost_cloud
/move_base/TrajectoryPlannerROS/global_plan
/move_base/TrajectoryPlannerROS/local_plan
/move_base/TrajectoryPlannerROS/parameter_descriptions
/move_base/TrajectoryPlannerROS/parameter_updates
/move_base/cancel
/move_base/current_goal
/move_base/feedback
/move_base/global_costmap/costmap
/move_base/global_costmap/costmap_updates
/move_base/global_costmap/footprint
/move_base/global_costmap/inflater_layer/parameter_descriptions
/move_base/global_costmap/inflater_layer/parameter_updates
/move_base/global_costmap/obstacles_layer/parameter_descriptions
/move_base/global_costmap/obstacles_layer/parameter_updates
/move_base/global_costmap/parameter_descriptions
/move_base/global_costmap/parameter_updates
/move_base/global_costmap/static_layer/parameter_descriptions
/move_base/global_costmap/static_layer/parameter_updates
/move_base/goal
/move_base/local_costmap/costmap
/move_base/local_costmap/costmap_updates
/move_base/local_costmap/footprint
/move_base/local_costmap/inflater_layer/parameter_descriptions
/move_base/local_costmap/inflater_layer/parameter_updates
/move_base/local_costmap/obstacles_layer/parameter_descriptions
/move_base/local_costmap/obstacles_layer/parameter_updates
/move_base/local_costmap/parameter_descriptions
/move_base/local_costmap/parameter_updates
/move_base/local_costmap/static_layer/parameter_descriptions
/move_base/local_costmap/static_layer/parameter_updates
/move_base/parameter_descriptions
/move_base/parameter_updates
/move_base/result
/move_base/status
/move_base_simple/goal
/obj_error
/odom
/odometry/filtered
/particlecloud
/power_load
/ridgeback_velocity_controller/cmd_vel
/ridgeback_velocity_controller/odom
/rosgraph_manipulator_action_server/cancel
/rosgraph_manipulator_action_server/feedback
/rosgraph_manipulator_action_server/goal
/rosgraph_manipulator_action_server/result
/rosgraph_manipulator_action_server/status
/rosout
/rosout_agg
/scan_filtered
/set_pose
/tf
/tf_static

```