meta-control/mc_mros_reasoner #83: [WARN] [1599118979.033952]: Unsupported QA TYPE received: objetive_error

chcorbato, 2020-09-03:
commit: df76904bb6e2b5f78eefe65c5e406ed6efc0f454 (melodic branch)

test MODELS20:
`./run_single_sim.sh -p 1 -g 1 -n "f1_v2_r2" -o 2 -r "true" -p 1.7 -c "false"
`

gavanderhoorn, 2020-09-03:
Suggestion: include a verbatim copy-paste of "all" output, not just a single line.

Context can be important when trying to diagnose these sort of issues.

chcorbato, 2020-09-03:
Thanks for the reminder. I normally do, but I have so many prints in `mros_reasoner.py` that it is a mess. Tthis corresponds to a feature not implemented, so this is simply a reminder for me.

marioney, 2020-09-03:
I think this error comes from an additional observer we added before.  It's solved in https://github.com/rosin-project/metacontrol_sim/commit/9fe54db8af7ac01005fc74e09155eb85e9454ecd

chcorbato, 2020-09-03:
> I think this error comes from an additional observer we added before. It's solved in [rosin-project/metacontrol_sim@9fe54db](https://github.com/rosin-project/metacontrol_sim/commit/9fe54db8af7ac01005fc74e09155eb85e9454ecd)

That is the case. Thanks @marioney

closed 2020-09-03 by chcorbato