meta-control/mc_mros_reasoner #89: Sometimes diagnostic msgs with QA values are not received by reasoner

chcorbato, 2020-09-08:
This is the observed behavior, the FG QAvalues remains empty [] in the reasoner output during the execution of:
` ./run_single_sim.sh -p 1 -g 1 -n "f1_v2_r2" -o 2 -r "true" -p 1.7 -c "false"
`

```
[INFO] [1599558522.641155]:      >> Finished ontological reasoning)

FGs:
fg_f1_v2_r2 	objective:  kb.o_navigateA 	status:  None 	FD:  kb.f1_v2_r2 	QAvalues:  []

OBJECTIVE	|  STATUS	|  NFRs
o_navigateA 	|   None 	|   [('energy', 0.7), ('safety', 0.4)]

```

chcorbato, 2020-09-08:
This happens to me very often, but I have not been able to figure out how to reproduce it.
Does this happen to you too @marioney ?

chcorbato, 2020-09-09:
The problem seem to be in the messages tat should be published in `/diagnostics`:
```
rostopic echo /diagnostics
WARNING: no messages received and simulated time is active.
Is /clock being published?
```

chcorbato, 2020-09-09:
Does this happens to you @JWijkhuizen ?

JWijkhuizen, 2020-09-16:
I do not have this problems with the Observers. But I am still running most of my simulations without the reasoner

JWijkhuizen, 2020-09-17:
> The problem seem to be in the messages tat should be published in `/diagnostics`:
> 
> ```
> rostopic echo /diagnostics
> WARNING: no messages received and simulated time is active.
> Is /clock being published?
> ```

This indicates that it is a problem with the observer right?

I have had the problem that some of the observers stopped publishing after move_base was killed and launched again. For now I fixed this by also killing and launching the observers as well, which is fine for my simulation where I only collect data. 

I think the problem is in the rosgraph_monitor node

chcorbato, 2020-09-17:
Thanks for the insights @JWijkhuizen , I will look into it.