meta-control/mc_mros_reasoner #93: Failed build test_catkin_virtualenv in branch vritual_env (#85)

chcorbato, 2020-09-09:
(in relation to #85 )
Following the instructions in https://github.com/tud-cor/mc_mros_reasoner/blob/virtual_env/README.md
I get this error:
```
Starting  >>> test_catkin_virtualenv                                           
_______________________________________________________________________________
Warnings   << test_catkin_virtualenv:cmake /home/parallels/tests_mros_reasoner_ws/logs/test_catkin_virtualenv/build.cmake.000.log
Traceback (most recent call last):
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/scripts/collect_requirements", line 23, in <module>
    from catkin_virtualenv import configure_logging, collect_requirements
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/collect_requirements.py", line 25, in <module>
    from catkin.find_in_workspaces import find_in_workspaces
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/find_in_workspaces.py", line 39, in <module>
    from catkin_pkg.packages import find_packages
ModuleNotFoundError: No module named 'catkin_pkg'
Traceback (most recent call last):
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/scripts/collect_requirements", line 23, in <module>
    from catkin_virtualenv import configure_logging, collect_requirements
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/collect_requirements.py", line 25, in <module>
    from catkin.find_in_workspaces import find_in_workspaces
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/find_in_workspaces.py", line 39, in <module>
    from catkin_pkg.packages import find_packages
ModuleNotFoundError: No module named 'catkin_pkg'
cd /home/parallels/tests_mros_reasoner_ws/build/test_catkin_virtualenv; catkin build --get-env test_catkin_virtualenv | catkin env -si  /usr/bin/cmake /home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/test_catkin_virtualenv --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/parallels/tests_mros_reasoner_ws/devel/.private/test_catkin_virtualenv -DCMAKE_INSTALL_PREFIX=/home/parallels/tests_mros_reasoner_ws/install; cd -
...............................................................................
_______________________________________________________________________________
Errors     << test_catkin_virtualenv:make /home/parallels/tests_mros_reasoner_ws/logs/test_catkin_virtualenv/build.make.000.log
Traceback (most recent call last):
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/scripts/venv_init", line 24, in <module>
    from catkin_virtualenv.venv import Virtualenv
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/venv.py", line 33, in <module>
    from .collect_requirements import collect_requirements
  File "/home/parallels/tests_mros_reasoner_ws/src/catkin_virtualenv/catkin_virtualenv/src/catkin_virtualenv/collect_requirements.py", line 25, in <module>
    from catkin.find_in_workspaces import find_in_workspaces
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/find_in_workspaces.py", line 39, in <module>
    from catkin_pkg.packages import find_packages
ModuleNotFoundError: No module named 'catkin_pkg'
make[2]: *** [venv/bin/python] Error 1
make[1]: *** [CMakeFiles/test_catkin_virtualenv_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/parallels/tests_mros_reasoner_ws/build/test_catkin_virtualenv; catkin build --get-env test_catkin_virtualenv | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << test_catkin_virtualenv:make                        [ Exited with code 2 ]
Failed    <<< test_catkin_virtualenv                             [ 2.7 seconds ]
Abandoned <<< test_catkin_virtualenv_py3                         [ Unrelated job failed ]
Abandoned <<< test_catkin_virtualenv_py3_isolated                [ Unrelated job failed ]
Abandoned <<< mros1_reasoner                                     [ Unrelated job failed ]
Abandoned <<< test_catkin_virtualenv_inherited                   [ Unrelated job failed ]
Finished  <<< metacontrol_msgs                                   [ 6.8 seconds ]
[build] Summary: 4 of 9 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: 4 packages were abandoned.                                
[build]   Failed:    1 packages failed.                                        
[build] Runtime: 10.5 seconds total.                                           
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

chcorbato, 2020-09-09:
Update on potentially missing dependencies:

```
rosdep check --from-paths src/ --ignore-src
System dependencies have not been satisfied:
apt	virtualenv
apt	python3-rospkg-modules
apt	python3-nose
```
and installing dependencies:
```
rosdep install --from-paths src/ --ignore-src
executing command [sudo -H apt-get install virtualenv]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.15.0-34 linux-headers-4.15.0-34-generic linux-image-4.15.0-34-generic
  linux-modules-4.15.0-34-generic linux-modules-extra-4.15.0-34-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  python3-virtualenv
The following NEW packages will be installed:
  python3-virtualenv virtualenv
0 upgraded, 2 newly installed, 0 to remove and 299 not upgraded.
Need to get 47.8 kB of archives.
After this operation, 171 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 python3-virtualenv all 15.1.0+ds-1.1 [43.4 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu bionic/universe amd64 virtualenv all 15.1.0+ds-1.1 [4,476 B]
Fetched 47.8 kB in 1s (92.5 kB/s)     
Selecting previously unselected package python3-virtualenv.
(Reading database ... 278666 files and directories currently installed.)
Preparing to unpack .../python3-virtualenv_15.1.0+ds-1.1_all.deb ...
Unpacking python3-virtualenv (15.1.0+ds-1.1) ...
Selecting previously unselected package virtualenv.
Preparing to unpack .../virtualenv_15.1.0+ds-1.1_all.deb ...
Unpacking virtualenv (15.1.0+ds-1.1) ...
Setting up python3-virtualenv (15.1.0+ds-1.1) ...
Setting up virtualenv (15.1.0+ds-1.1) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...
executing command [sudo -H apt-get install python3-rospkg-modules]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  linux-headers-4.15.0-34 linux-headers-4.15.0-34-generic linux-image-4.15.0-34-generic
  linux-modules-4.15.0-34-generic linux-modules-extra-4.15.0-34-generic
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  python3-catkin-pkg-modules python3-docutils python3-pygments python3-pyparsing python3-roman
Suggested packages:
  fonts-linuxlibertine | ttf-linux-libertine texlive-lang-french texlive-latex-base texlive-latex-recommended
  python-pyparsing-doc
The following NEW packages will be installed:
  python3-catkin-pkg-modules python3-docutils python3-pygments python3-pyparsing python3-roman
  python3-rospkg-modules
0 upgraded, 6 newly installed, 0 to remove and 299 not upgraded.
Need to get 1,064 kB of archives.
After this operation, 5,686 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
Get:1 http://ftp.tudelft.nl/ros/ubuntu bionic/main amd64 python3-catkin-pkg-modules all 0.4.22-1 [42.2 kB]
Get:2 http://ftp.tudelft.nl/ros/ubuntu bionic/main amd64 python3-rospkg-modules all 1.2.8-1 [23.9 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 python3-roman all 2.0.0-3 [8,624 B]
Get:4 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 python3-docutils all 0.14+dfsg-3 [363 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 python3-pyparsing all 2.2.0+dfsg1-2 [52.2 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu bionic/main amd64 python3-pygments all 2.2.0+dfsg-1 [574 kB]
Fetched 1,064 kB in 1s (1,041 kB/s)     
Selecting previously unselected package python3-roman.
(Reading database ... 278681 files and directories currently installed.)
[... 58 lines cut ...]
launchpadlib 1.10.6 requires testresources, which is not installed.
usage: venv_install [-h] --requirements REQUIREMENTS [REQUIREMENTS ...]
                    [--extra-pip-args EXTRA_PIP_ARGS]
                    venv
venv_install: error: argument --requirements: expected at least one argument
make[2]: *** [venv/bin/activate] Error 2
make[2]: *** Deleting file 'venv/bin/activate'
make[1]: *** [CMakeFiles/test_catkin_virtualenv_generate_virtualenv.dir/all] Error 2
make: *** [all] Error 2
cd /home/parallels/tests_mros_reasoner_ws/build/test_catkin_virtualenv; catkin build --get-env test_catkin_virtualenv | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Failed     << test_catkin_virtualenv:make                        [ Exited with code 2 ]                           
Failed    <<< test_catkin_virtualenv                             [ 6.9 seconds ]                                  
Abandoned <<< test_catkin_virtualenv_py3_isolated                [ Unrelated job failed ]                         
Abandoned <<< mros1_reasoner                                     [ Unrelated job failed ]                         
Abandoned <<< test_catkin_virtualenv_inherited                   [ Unrelated job failed ]                         
Finished  <<< test_catkin_virtualenv_py3                         [ 11.7 seconds ]                                 
[build] Summary: 4 of 8 packages succeeded.                                                                       
[build]   Ignored:   None.                                                                                        
[... cut ...]                                                               
[build]   Abandoned: 3 packages were abandoned.                                                                   
[build]   Failed:    1 packages failed.                                                                           
[build] Runtime: 13.3 seconds total.                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

chcorbato, 2020-09-09:
> launchpadlib 1.10.6 requires testresources, which is not installed.

Maybe this is the problem @marioney ?

chcorbato, 2020-09-09:
after 
`sudo apt install python3-testresources`

I have compiled the ws with 2 warnings on `test_catkin_virtualenv` and `test_catkin_virtualenv_inherited`:
```
Warnings   << test_catkin_virtualenv:make /home/parallels/tests_mros_reasoner_ws/logs/test_catkin_virtualenv/build.make.000.log
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

launchpadlib 1.10.6 requires testresources, which is not installed.
cd /home/parallels/tests_mros_reasoner_ws/build/test_catkin_virtualenv; catkin build --get-env test_catkin_virtualenv | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Finished  <<< test_catkin_virtualenv                             [ 12.3 seconds ]                                 
Starting  >>> test_catkin_virtualenv_py3_isolated                                                                 
Finished  <<< test_catkin_virtualenv_py3                         [ 9.6 seconds ]                                  
Starting  >>> mros1_reasoner                                                                                      
Finished  <<< mros1_reasoner                                     [ 1.7 seconds ]                                  
Starting  >>> test_catkin_virtualenv_inherited                                                                    
Finished  <<< test_catkin_virtualenv_py3_isolated                [ 8.4 seconds ]                                  
__________________________________________________________________________________________________________________
Warnings   << test_catkin_virtualenv_inherited:make /home/parallels/tests_mros_reasoner_ws/logs/test_catkin_virtualenv_inherited/build.make.000.log
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

launchpadlib 1.10.6 requires testresources, which is not installed.
cd /home/parallels/tests_mros_reasoner_ws/build/test_catkin_virtualenv_inherited; catkin build --get-env test_catkin_virtualenv_inherited | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..................................................................................................................
Finished  <<< test_catkin_virtualenv_inherited                   [ 11.5 seconds ]                                 
[build] Summary: All 9 packages succeeded!                                                                        
[build]   Ignored:   None.                                                                                        
[build]   Warnings:  2 packages succeeded with warnings.                                                          
[build]   Abandoned: None.                                                                                        
[build]   Failed:    None.                                                                                        
[build] Runtime: 35.6 seconds total.                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

gavanderhoorn, 2020-09-09:
I don't believe @chcorbato intended for this to get closed when #85 was merged.

@chcorbato: *linking* a PR will close the linked issue when the PR gets merged.

marioney, 2020-09-09:
I've tested this

```
wstool init ~/mros1_reasoner_ws/src https://raw.githubusercontent.com/tud-cor/mc_mros_reasoner/master/mros1_reasoner/mros1_reasoner.rosinstall

rosdep install --from-paths ~/mros1_reasoner_ws/src -y -i -r
```

On a clean docker image and it works well, no errors or nothing.

 I think adding the `--ignore-src` in

> rosdep install --from-paths src/ --ignore-src

 may cause some dependencies not to be installed

gavanderhoorn, 2020-09-09:
> I think adding the `--ignore-src` in
> 
> > rosdep install --from-paths src/ --ignore-src
> 
> may cause some dependencies not to be installed

`-i` is the short form of `--ignore-src`.

chcorbato, 2020-09-09:
@JWijkhuizen and/or @shreyashpalande, could you try to install `mc_mros_reasoner:master` and confirm whether you do not run into this issue?

shreyashpalande, 2020-09-09:
@chcorbato I was able to run `mc_mros_reasoner:master` branch successfully without any issues. This is the output 

started roslaunch server 

```
SUMMARY
========

PARAMETERS
 * /onto_file: /home/palle/mros1...
 * /rosdistro: melodic
 * /rosversion: 1.14.9

NODES
  /
    reasoner (mros1_reasoner/mros1_reasoner.py)

auto-starting new master
process[master]: started with pid [23281]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ed951d4a-f2ae-11ea-86cd-c0b6f95b09e1
process[rosout-1]: started with pid [23295]
started core service [/rosout]
process[reasoner-2]: started with pid [23302]
* Owlready2 * Warning: optimized Cython parser module 'owlready2_optimized' is not available, defaulting to slower Python implementation
[INFO] [1599664372.112699]: Loaded ontology: /home/palle/mros1_reasoner_ws/src/mros1_reasoner/mros1_reasoner/scripts/kb.owl
[WARN] [1599664372.113976]: grounded_configuration not found in the param server
[INFO] [1599664372.114588]: KB initialization:
 	 - Supported QAs: 
 	 	 - for Function f_navigate: /nfr_energy, /nfr_safety 
 	 - If an Objective instance is not found in the owl file, a default o_navigate is created.
[INFO] [1599664372.115417]: Creating default Objective o_navigateA with default NFRs
[WARN] [1599664372.117177]: No value in rosparam server for /nfr_energy, setting it to 0.5
[WARN] [1599664372.119053]: No value in rosparam server for /nfr_safety, setting it to 0.8
[INFO] [1599664374.128111]: Entered timer_cb for metacontrol reasoning
[INFO] [1599664374.131524]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
* Owlready2 * Running Pellet...
    java -Xmx2000M -cp /home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-tdb-0.10.0.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-iri-0.9.5.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-log4j12-1.6.4.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-runtime-3.2.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/log4j-1.2.16.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jgrapht-jdk1.5.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-core-2.10.0.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/pellet-2.3.1.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/aterm-java-1.6.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-api-1.6.4.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jcl-over-slf4j-1.6.4.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpclient-4.2.3.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/owlapi-distribution-3.4.3-bin.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-arq-2.10.0.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/commons-codec-1.6.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-3.2.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xml-apis-1.4.01.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xercesImpl-2.10.0.jar:/home/palle/mros1_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpcore-4.2.2.jar pellet.Pellet realize --loader Jena --input-format N-Triples --infer-prop-values --infer-data-prop-values --ignore-imports /tmp/tmpcca9s7ta
* Owlready2 * Pellet took 0.7029156684875488 seconds
* Owlready * (NB: only changes on entities loaded in Python are shown, other changes are done but not listed)
[INFO] [1599664374.862068]:      >> Finished ontological reasoning)

FGs:
fg_None 	objective:  kb.o_navigateA 	status:  None 	FD:  None 	QAvalues:  []

OBJECTIVE	|  STATUS	|  NFRs
o_navigateA 	|   None 	|   [('energy', 0.5), ('safety', 0.8)]
[INFO] [1599664374.867218]: No Objectives in status ERROR: no adaptation is needed
[INFO] [1599664374.869447]:   >> Finished MAPE-K ** ANALYSIS **
[INFO] [1599664374.871586]:   >> Started MAPE-K ** PLAN adaptation **
[INFO] [1599664374.873486]: - NO ADAPTATION NEEDED -
[INFO] [1599664374.874976]: Exited timer_cb for metacontrol reasoning
[INFO] [1599664376.127423]: Entered timer_cb for metacontrol reasoning
[INFO] [1599664376.130639]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
```

chcorbato, 2020-09-11:
Based on this I close the [test above](https://github.com/tud-cor/mc_mros_reasoner/issues/93#issuecomment-689630119) by @shreyashpalande  issue

closed 2020-09-11 by chcorbato