meta-control/mc_mros_reasoner #101: Sometimes test_level_1_functional_arch fails

chcorbato, 2020-09-18:
commit: ef4d6f039139a6ef162a7ac71228a8617dff5020

In some failed tests, there is a reasoning ERROR reported in rosout, see:
```console
[INFO] [1600433487.836296]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
* Owlready2 * Running Pellet...
    java -Xmx2000M -cp /home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-log4j12-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpcore-4.2.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xercesImpl-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jcl-over-slf4j-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-3.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-api-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpclient-4.2.3.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-core-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-tdb-0.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jgrapht-jdk1.5.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/pellet-2.3.1.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-runtime-3.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-iri-0.9.5.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/owlapi-distribution-3.4.3-bin.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/aterm-java-1.6.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xml-apis-1.4.01.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/commons-codec-1.6.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-arq-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/log4j-1.2.16.jar pellet.Pellet realize --loader Jena --input-format N-Triples --infer-prop-values --infer-data-prop-values --ignore-imports /tmp/tmp0567lvdy
[ERROR] [1600433487.851572]: Reasoning error

FGs:
fg_print        objective:  test1.o_print       status:  None   FD:  test1.fd_print_1   QAvalues:  []

OBJECTIVE       |  STATUS       |  NFRs
o_print         |   None        |   []
[INFO] [1600433487.853534]: No Objectives in status ERROR: no adaptation is needed
[INFO] [1600433487.854436]:   >> Finished MAPE-K ** ANALYSIS **
[INFO] [1600433487.855391]: Exited timer_cb for metacontrol reasoning
[Testcase: testtest_level_1_functional_arch] ... ok

[ROSTEST]-----------------------------------------------------------------------

[mros1_reasoner.rosunit-test_level_1_functional_arch/test_1_level_functional_architecture][FAILURE]
False is not true
  File "/usr/lib/python3.6/unittest/case.py", line 59, in testPartExecutor
    yield
  File "/usr/lib/python3.6/unittest/case.py", line 605, in run
    testMethod()
  File "/home/parallels/tests_mros_reasoner_ws/src/mros1_reasoner/mros1_reasoner/test/test_level_1_functional_arch.py", line 74, in test_1_level_functional_architecture
    self.assertTrue(self.success)
  File "/usr/lib/python3.6/unittest/case.py", line 682, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[mros1_reasoner.rosunit-test_level_1_functional_arch/test_one_equals_one][passed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 2
 * ERRORS: 0
 * FAILURES: 1
```

but in other failed tests there is not such error reported:
```console
[INFO] [1600433505.407305]:   >> Started MAPE-K ** Analysis (ontological reasoning) **
* Owlready2 * Running Pellet...
    java -Xmx2000M -cp /home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-log4j12-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpcore-4.2.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xercesImpl-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jcl-over-slf4j-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-3.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/slf4j-api-1.6.4.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/httpclient-4.2.3.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-core-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-tdb-0.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jgrapht-jdk1.5.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/pellet-2.3.1.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/antlr-runtime-3.2.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-iri-0.9.5.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/owlapi-distribution-3.4.3-bin.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/aterm-java-1.6.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/xml-apis-1.4.01.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/commons-codec-1.6.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/jena-arq-2.10.0.jar:/home/parallels/tests_mros_reasoner_ws/devel/.private/mros1_reasoner/share/mros1_reasoner/venv/lib/python3.6/site-packages/owlready2/pellet/log4j-1.2.16.jar pellet.Pellet realize --loader Jena --input-format N-Triples --infer-prop-values --infer-data-prop-values --ignore-imports /tmp/tmpnhp3b4fe
[Testcase: testtest_level_1_functional_arch] ... ok

[ROSTEST]-----------------------------------------------------------------------

[... cut ...]
  File "/home/parallels/tests_mros_reasoner_ws/src/mros1_reasoner/mros1_reasoner/test/test_level_1_functional_arch.py", line 74, in test_1_level_functional_architecture
    self.assertTrue(self.success)
  File "/usr/lib/python3.6/unittest/case.py", line 682, in assertTrue
    raise self.failureException(msg)
--------------------------------------------------------------------------------

[mros1_reasoner.rosunit-test_level_1_functional_arch/test_one_equals_one][passed]

SUMMARY
 * RESULT: FAIL
 * TESTS: 2
 * ERRORS: 0
 * FAILURES: 1
```

marioney, 2020-09-18:
Update:

It seems that the error happens when the [`timer_cb`](https://github.com/tud-cor/mc_mros_reasoner/blob/ef4d6f039139a6ef162a7ac71228a8617dff5020/mros1_reasoner/src/mros1_reasoner/ros_reasoner.py#L167)  is called during the test.

chcorbato, 2020-09-18:
> Update:
> 
> It seems that the error happens when the [`timer_cb`](https://github.com/tud-cor/mc_mros_reasoner/blob/ef4d6f039139a6ef162a7ac71228a8617dff5020/mros1_reasoner/src/mros1_reasoner/ros_reasoner.py#L167) is called during the test.

I do not know how rostest execution works: how would it be possible not to call `timer_cb` during the test? It is a sync we want to test.

marioney, 2020-09-21:
Ok, found the error.

This message

https://github.com/tud-cor/mc_mros_reasoner/blob/aace2aeb04854b373c16b40c82b9451fc58311b9/mros1_reasoner/src/mros1_reasoner/ros_reasoner.py#L54

Is sometimes published before the tests subscribes to the topic, and therefore it's always failing.

@chcorbato  we can solve this in two ways:

[ ] Add a sleep before publishing the message.
[ ] Use a different message (one that is published afterwards, i.e in the timer)

chcorbato, 2020-09-22:
Shall we close the issue @marioney ? Unless you have a better fix already in mind, I propose you open a different issue if it needs to be fixed some other way and we close this one (not to have this one hanging if we do not have to)

marioney, 2020-09-22:
Ok, we should close this then. It's ok to leave like this I guess.

closed 2020-09-22 by marioney