meta-control/mc_mros_reasoner #106: Create ontology for laser_failure according to system modes in pilot URJC

marioney, 2020-09-24:
We new an ontology that includes the names of the system modes of https://github.com/MROS-RobMoSys-ITP/Pilot-URJC so as to be able to test the reasoner with the pilot

@jginesclavero can maybe help us with this

jginesclavero, 2020-09-25:
Hi @marioney!
Of course, How can I help you?

chcorbato, 2020-11-10:
I think this has been completed by @marioney and @estherag, could you please indicate where?

estherag, 2020-11-11:
I think [here](https://github.com/tud-cor/mc_mros_reasoner/blob/foxy_devel/mros2_reasoner/owl/MROS_ontology.owl) is the ontology @marioney has updated to use `laser_resender` individual rather than `laser` to match modes configuration file.

However, for future components, to avoid information duplicates and mismatching names, maybe we could read this configuration file and update MROS ontology individuals with an owlready script?

marioney, 2020-11-11:
> I think [here](https://github.com/tud-cor/mc_mros_reasoner/blob/foxy_devel/mros2_reasoner/owl/MROS_ontology.owl) is the ontology @marioney has updated to use `laser_resender` individual rather than `laser` to match modes configuration file.

That's correct.  That version of the ontology already handles the `laser_resender` error.

> However, for future components, to avoid information duplicates and mismatching names, maybe we could read this configuration file and update MROS ontology individuals with an owlready script?

@estherag  Not sure what you mean by that? you mean to create the components from the python script? that will require to also change all the `fd_s` to determine which ones depend on it, is that correct?

estherag, 2020-11-13:
@marioney that's it. However, this can be done through that python script easily. Just like the metacontroller currently creates an objective and its NFRs and links it to be their QA. We could have all the `fd_s` in the .owl and in the python script define the components and link them to the `fd_s` through the property `requiredBy`. Just like I did manually when I created the ontology.

I just proposed it for re-use information from the system modes YAML and avoid naming errors, but maybe is not worth it because the aren't so many components.

marioney, 2020-11-15:
@estherag I've created a separated issue for that (see #112)

I'm closing this one as the laser_failure is already handled by the current ontology.

closed 2020-11-15 by marioney