meta-control/mc_mros_reasoner #111: Cancel goal when no available fd is found

marioney, 2020-11-10:
If the objective is in error, and the Metacontroller does not find any valid mode to solve it, the goal should be cancelled.

@chcorbato is this the correct way to go? should it be an `fd_error` to go when all other "modes" are not avaible

chcorbato, 2020-11-16:
> If the objective is in error, and the Metacontroller does not find any valid mode to solve it, the goal should be cancelled.

I agree, the objective for the `mros_reasoner` should be cancelled. I think this would require:
- [ ] remove the objective instance from those the reasoner is trying to maintain/pursue. Maybe this requires to extend in the ontology the status of an objective. What do you think @estherag @rsanz ?
- [ ] report back, to the client of the `mros_reasoner` metacontrol services, that the objective cannot be solved by `mros_reasoner`. Something for the interface to be discussed with URJC

> @chcorbato is this the correct way to go? should it be an `fd_error` to go when all other "modes" are not available

This is an interesting idea. To define a safe mode/configuration/architecture when the system is in error. But note that the fact that an objective is in error, i.e. cannot be achieved does not mean that the system is in error. Maybe it is simply out of the scope of the available FDs.

I think the current definition of the `status` property of an `Objective` in the ontology is not expressive enough for all these cases.

estherag, 2020-11-17:
Maybe we can have more status for the objective: IN_PROGRESS, IN_ERROR_NFR, IN_ERROR_COMPONENT, UNREACHABLE.

IN_ERROR_NFR may be set when the current FG doesn't meet objectives NFRs so a new FG should be selected
IN_ERROR_COMPONENT would be set when the current FG uses an unavailable component so a new FG should be selected
When there isn't a FD reachable, we could set the objective at UNREACHABLE, then we can delete this objective and switch to another one (i.e. switch from navigating to go to recharge). In this case, maybe we can miss NFR because is an emergency action and navigate to the deck with fewer restrictions so FD with IN_ERROR_NFR property on the log can be selected.

chcorbato, 2020-11-17:
I like your proposal @estherag 

In the case of UNREACHABLE, I think the metacontroller should notify the client of its action server to achieve that objective, typically the task/mission coordination logic, who would decide whether to cancel that objective (thus the reasoner would eliminate it from the ontology) and which new objective to request.

We have discussed with URJC and agreed that the logic and decision to instantiate a new objective to navigate to the recharge station should not me in the reasoner/metacontroller, but in the application/task/mission logic.
The reason is to make the metacontroller as independent from the application (e.g. navigation) and reusable as possible.
I hope you also agree @estherag @rsanz

marioney, 2020-11-17:
I agree, having more status makes sense.  

> In the case of UNREACHABLE, I think the metacontroller should notify the client of its action server to achieve that objective, typically the task/mission coordination logic, who would decide [...]

Still, shoudl the ontology define a  FG to choose when the objective is UNREACHABLE? or should it just report that it cannot be achieved and let the processing for the metacontroller/task planner?

estherag, 2020-11-23:
I also agree.

I think from the ontology side both approach are possible:

I understand this UNREACHABLE status is set by the metacontroller when there is not FD available. In that case, the mission logic should set a new objective.  I think we could go in this case with the objective and FG in error until a new objective and FG are set. 

However, if it is easier for the metacontroller to have a `processing_unreachable_objective`, we could have a FD in unavailable status that will become available when the objective is UNREACHABLE. Once this FD is available, we can make its corresponding FG.

chcorbato, 2020-11-24:
I have moved this question of having a fall-back or default FD/configuration/mode to a new issue #118 
I close this one, since certainly the objective being UNREACHABLE should be reported by the reasoner to the wrapper and then to the mission logic.

closed 2020-11-24 by chcorbato