meta-control/mc_mros_reasoner #113: [ros2] Read more than two ontologies

norro, 2020-11-19:
Currently, the mros2 reasoner setup seems to assume that I pass one model fle (my custom ontology, I suppose) and the tomasys ontology. What would be the recommended way of loading tomasys, and two more ontology files (the two MROS pilot ontologies in my case)?

norro, 2020-11-19:
@chcorbato @marioney Can you help with this issue?

marioney, 2020-11-19:
I've created a branch to test this https://github.com/tud-cor/mc_mros_reasoner/tree/foxy_multi_model_files

It accepts a list of model files in the [launch file](https://github.com/tud-cor/mc_mros_reasoner/blob/1fc640c552cac0fae5ef63434be7d5107d4a5881/mros2_reasoner/launch/launch_reasoner.launch.py#L40) and appends the secondary model files to the first one. 

@chcorbato  do you think this will work?

Still we need to test if it gets correct "reasoning" using the two files.  I'm getting some errors when loading the bosh ontology, so I'm not being able to test it.

I've created an issue in https://github.com/MROS-RobMoSys-ITP/metacontrol_ontology_bosch/issues/2 to have that as a ros package and facilitate testing.

marioney, 2020-11-23:
@norro Can you confirm if this is working, so we can close this and make a PR with the changes?

norro, 2020-11-23:
This is working for us, thank you, @marioney!

closed 2020-11-23 by norro