meta-control/mc_mros_reasoner #114: mc_mros_reasoner package release

chcorbato, 2020-11-23:
(from https://github.com/MROS-RobMoSys-ITP/metacontroller/issues/4)
metacontroller depends on owlready2 python package
-[ ] we need to contribute the rule to rosdep:
https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md
http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html

gavanderhoorn, 2020-11-23:
Note: since `owlready2` is not available through the Debian/Ubuntu `apt` repositories, you'll have to contribute a `pip` rule (ie: `pip-owlready2`).

This will immediately mean you cannot release `mc_mros_reasoner` through the ROS buildfarm (as that only supports `apt` dependencies), but I guess that's not really an issue here.

marioney, 2020-11-23:
I've created the PR to contribute the `python3-owlready2-pip` rule  https://github.com/ros/rosdistro/pull/27418

> Edit: and actually, if you're still using `catkin_virtualenv`, I believe adding the dependency to `rosdep` is not necessary. `pip` (via `catkin_virtualenv`) will take care of this.

We're using both, `catkin_virtualenv` is used for melodic, but for foxy we don't need that anymore.

gavanderhoorn, 2020-11-23:
> We're using both, `catkin_virtualenv` is used for melodic, but for foxy we don't need that anymore.

could still be nice to avoid polluting people's machines, but that may be not too important.

marioney, 2020-11-23:
> could still be nice to avoid polluting people's machines, but that may be not too important.

In any case, I guess the `rosdep` rule is still useful.

I'm closing this as the contribution has been merged already https://github.com/ros/rosdistro/pull/27418#event-4029472094

closed 2020-11-23 by marioney