meta-control/mc_mros_reasoner #130: moduleNotFound: No module named “system_modes”

FahimaGold, 2022-11-10:
Hello, I am a beginner in ROS, and I am working on MROS. I followed the installation guidelines in README, and using the branch `foxy_devel.` However, when I run `ros2 launch mros2_reasoner launch_reasoner.launch.py`. I got the following error:

> vagrant@ubuntu-vb:~/mros_reasoner_ws$ ros2 launch mros2_reasoner launch_reasoner.launch.py
[INFO] [launch]: All log files can be found below /home/vagrant/.ros/log/2022-11-10-10-35-18-087379-ubuntu-vb-72166
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [mros2_reasoner_node-1]: process started with pid [72168]
[INFO] [mros2_wrapper-2]: process started with pid [72170]
[mros2_reasoner_node-1] Traceback (most recent call last):
[mros2_reasoner_node-1]   File "/home/vagrant/mros_reasoner_ws/install/mros2_reasoner/lib/mros2_reasoner/mros2_reasoner_node", line 11, in <module>
[mros2_reasoner_node-1]     load_entry_point('mros2-reasoner', 'console_scripts', 'mros2_reasoner_node')()
[mros2_reasoner_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 490, in load_entry_point
[mros2_reasoner_node-1]     return get_distribution(dist).load_entry_point(group, name)
[mros2_reasoner_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2854, in load_entry_point
[mros2_reasoner_node-1]     return ep.load()
[mros2_reasoner_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2445, in load
[mros2_reasoner_node-1]     return self.resolve()
[mros2_reasoner_node-1]   File "/usr/lib/python3/dist-packages/pkg_resources/__init__.py", line 2451, in resolve
[mros2_reasoner_node-1]     module = __import__(self.module_name, fromlist=['__name__'], level=0)
[mros2_reasoner_node-1]   File "/home/vagrant/mros_reasoner_ws/build/mros2_reasoner/mros2_reasoner/mros2_reasoner_node.py", line 2, in <module>
[mros2_reasoner_node-1]     from mros2_reasoner.ros_reasoner import RosReasoner
[mros2_reasoner_node-1]   File "/home/vagrant/mros_reasoner_ws/build/mros2_reasoner/mros2_reasoner/ros_reasoner.py", line 7, in <module>
[mros2_reasoner_node-1]     from system_modes.srv import ChangeMode
[mros2_reasoner_node-1] ModuleNotFoundError: No module named 'system_modes'
[ERROR] [mros2_reasoner_node-1]: process has died [pid 72168, exit code 1, cmd '/home/vagrant/mros_reasoner_ws/install/mros2_reasoner/lib/mros2_reasoner/mros2_reasoner_node --ros-args -r __node:=mros2_reasoner_node --params-file /tmp/launch_params_y_o39mtr'].

system_modes is already installed, it is located under install/.

Rezenders, 2022-11-14:
The ChangeMode service is not in the system_modes pkg, it is in the system_modes_msgs pkg. That is why you are getting this error. I opened a PR to fix this. While it is not accepted you can change that directly in your code.

FahimaGold, 2022-11-15:
> The ChangeMode service is not in the system_modes pkg, it is in the system_modes_msgs pkg. That is why you are getting this error. I opened a PR to fix this. While it is not accepted you can change that directly in your code.

It worked, thank you very much !

PR #132 (2022-11-15): system_modes -> system_modes_msgs
System mode messages are in the system_modes_msgs package. Resolves #130.
Also, remove ChangeMode import for the fake_reasoner.
closed 2022-11-15 by FahimaGold