meta-control/mc_mros_reasoner #135: Reorganize branches

Rezenders, 2022-11-15:
Delete unused branches. I think it would be nice to make the main branch of this repo a ros2 branch.

chcorbato, 2022-11-24:
Agreed. I propose:

- [ ] Delete all branches that are merged into: `master`, `foxy_devel` or `humble`
- [ ] Create a pull request from https://github.com/MRamirez25/mc_mros_reasoner (check if the fixes there are still relevant)

Rezenders, 2022-11-25:
What about the branches `melodic`, `rosmodel2owl`, `rosnavigationowl`, and `refactoring`?
`melodic` and `rosnavigationowl` are behind master and don't have any commits to be merged
`rosmodel2owl` and `refactoring`  have commits to be merged and are behind master

Rezenders, 2022-12-19:
> Create a pull request from https://github.com/MRamirez25/mc_mros_reasoner (check if the fixes there are still relevant)

I think this is not necessary.

> I think it would be nice to make the main branch of this repo a ros2 branch.

I can't change the repo settings. @chcorbato Can you make the humble branch the main one?

chcorbato, 2023-01-12:
@Rezenders I have given the maintainers team (you and Elvin) Maintain access to the repo. Let me know in case this is not enough to do the actions you suggest, I agree with them.

In any case the main branch should be `master` (or if you prefer rename it to `main`) and for the moment be pointing to the same commit than `humble`

closed 2023-01-16 by Rezenders