meta-control/mc_mros_reasoner #141: All function designs in error are added to fd_error_log

Rezenders, 2022-11-29:
Every time a function grounding violates a NFR the objective it is solving is added to the function design `fd_error_log`. 
This shouldn't be the case for all violations, in the smart home for example we don't add the FDs to an error log. In addition, it doesn't make sense to block an FD for a violating a non-functional requirement. 

Another issue I see is that the relationship order seems to be inverted. I would say that an FD is added to the objective error log and not the opposite. 

But anyway, regarding the main issue I suggest:

- Making a distinction between non-functional requirements and functional requirements
- Only when a functional requirement is violated, the function design should be blocked

A quick solution for this in MROS could be to
- [ ] Add a property to quality attributes to indicate if it is a NFR or FR, by default I suggest treating them as NFR (this is not backwards compatible but I IMO it makes more sense)
- [ ] Add a new objective status called `IN_ERROR_FR`
- [ ] In the SWRL rules, have different rules for setting the objective status either to `IN_ERROR_FR` or `IN_ERROR_NFR` according to the quality attribute that was violated

Rezenders, 2022-11-29:
Done

Rezenders, 2022-11-29:
@chcorbato FYI

chcorbato, 2022-12-01:
We need to keep track of this change in the `tomasys` metamodel, i.e. values for 'Objective.status'
Have you edited tomasys.owl in some branch @Rezenders ?
Maybe good to indicate it here.

Rezenders, 2022-12-01:
I only modified mros.owl and its SWRL rules. Changes are in the main branch.
I didn't change the o_status, but it is working. I will add that to tomasys.

Rezenders, 2022-12-02:
Done at https://github.com/meta-control/mc_mdl_tomasys/commit/a020191e7e7c52d4f5aa3e2240fca2ea269bf940

closed 2022-12-02 by Rezenders