meta-control/mc_mros_reasoner #150: Define node that needs to be changed with a new configuration

jeroenzwan, 2022-12-06:
Currently the ros_reasoner doesn't take a node name into account for changing the system_mode of a node. This is supposed to happen in the request_configuration function, but this is not being done. One way to fix this for now is to maybe query the ontology for which components solve a fd, or these nodes need to be mapped directly to the fds. In the first case the node names would need to match the names of the components. As this is a bit of a conceptual decision I think it is good to maybe discuss this.

Rezenders, 2022-12-09:
Actually, this concept of node name is related to system modes, so I would suggest changing the `'/ros_reasoner/change_node_mode'` service to request a function design for a specific function. 

So, I suggest changing the service msg NodeMode to something like this:

```
string function_name
string required_function_design
---
```

Then in the system modes bridge we can have a logic to select the node to be changed. My suggestion is to use the function name as the default node name. Then, if we want to use a custom mapping, we could have a dict that maps functions and fds to different node names.

something like:

```python
function_design_mappings = {
  ('function', 'function_design'):'node1', 
  ('function', 'function_design2'):'node2'
}
```

then:

```python
if function_design_mappings has (req.function_name, req.required_function_design):
  node_name = function_design_mappins[(req.function_name, req.required_function_design)]
else:
  node_name =  req.function_name
```

jeroenzwan, 2022-12-13:
I pushed these changes to [this pull request](https://github.com/kas-lab/pipeline_inspection/pull/47). I think just using fd is enough, so I made a srv just with required_fd_name as argument.

Rezenders, 2023-01-04:
I will close this issue, since it was fixed by PR #160 by adding `system_modes_bridge.py`.

closed 2023-01-04 by Rezenders