meta-control/mc_mros_reasoner #156: Measured QAs are not saved unless there is a FG to attach it to

Rezenders, 2022-12-12:
Measured QAs are only saved if there is a FG to attach it to. Measured QAs are important even when there is no FG, see https://github.com/kas-lab/pipeline_inspection/issues/35.
In addition, the measured QAs are being saved attached to the first FG that is being found, unless in the diagnostic msg it is explicitly written what function it should be attached to. The user should not need to add this info to the diagnostics msg, the Fgs to attach it to should be inferred given the type of the QAs.

In tomasys.py
```python
def update_qa_value(fg, qa_type, value, tbox, abox):
    qas = fg.hasQAvalue

    if qas == []:  # for the first qa value received
        qav = tbox.QAvalue(
            "obs_{}".format(
                qa_type.name),
            namespace=abox,
            isQAtype=qa_type,
            hasValue=value)
        fg.hasQAvalue.append(qav)
    else:
        for qa in qas:
            if qa.isQAtype == qa_type:
                qa.hasValue = value
                return
        # case it is a new QA type value
        qav = tbox.QAvalue(
            "obs_{}".format(
                qa_type.name),
            isQAtype=qa_type,
            namespace=abox,
            hasValue=value)
        fg.hasQAvalue.append(qav)
```

In reasoner.py
```python
# update the QA value for an FG with the value received
def update_qa_value(fg, qa_type, value, tbox, abox):
    qas = fg.hasQAvalue

    if qas == []:  # for the first qa value received
        qav = tbox.QAvalue(
            "obs_{}".format(
                qa_type.name),
            namespace=abox,
            isQAtype=qa_type,
            hasValue=value)
        fg.hasQAvalue.append(qav)
    else:
        for qa in qas:
            if qa.isQAtype == qa_type:
                qa.hasValue = value
                return
        # case it is a new QA type value
        qav = tbox.QAvalue(
            "obs_{}".format(
                qa_type.name),
            isQAtype=qa_type,
            namespace=abox,
            hasValue=value)
        fg.hasQAvalue.append(qav)
```

Rezenders, 2022-12-12:
> Measured QAs are only saved if there is a FG to attach it to.

This is not true, they are saved with the prefix `obs_`. Like: `obs_mockiness`.

> In addition, the measured QAs are being saved attached to the first FG that is being found, unless in the diagnostic msg it is explicitly written what function it should be attached to. The user [...]

However, this seems to be true. I will check it later.

Rezenders, 2022-12-13:
> Measured QAs are only saved if there is a FG to attach it to.
This is not true, they are saved with the prefix obs_. Like: obs_mockiness.

Actually, this seems to be true.

chcorbato, 2023-01-12:
We need to define a consistent model for `QA values`, both in `tomasys` and the implement a corresponding logic in the reasoner.

I am reopening the issue for that reason @Rezenders , also the link to this discussion would be useful when addressing it.
But if you prefer to close this one and one a new Issue, that is ok with me.