meta-control/mc_mros_reasoner #162: Tomasys being used as abox, even though it is supposed to be tbox

jeroenzwan, 2023-01-04:
Instances of function designs are read from the tomasys file, which is supposed to be the tbox. However, it seems to me that the function design individuals should be part of the abox. I don't really understand how the the tomasys file even has access to this information, since this is part of the model file.

https://github.com/meta-control/mc_mros_reasoner/blob/7dd11c678ae0abc41459cb3123bb8d398c105acf/mros2_reasoner/mros2_reasoner/tomasys.py#L176-L183

chcorbato, 2023-01-12:
`tbox` is not the tomasys.owl file, but a Python object that is created in https://github.com/meta-control/mc_mros_reasoner/blob/7dd11c678ae0abc41459cb3123bb8d398c105acf/mros2_reasoner/mros2_reasoner/reasoner.py#L35 

https://github.com/meta-control/mc_mros_reasoner/blob/7dd11c678ae0abc41459cb3123bb8d398c105acf/mros2_reasoner/mros2_reasoner/tomasys.py#L35-L63
and
https://github.com/meta-control/mc_mros_reasoner/blob/7dd11c678ae0abc41459cb3123bb8d398c105acf/mros2_reasoner/mros2_reasoner/tomasys.py#L18-L32
probably with all the contents of both the ABox and TBox ol files

I think the issue here is a misleading name for the parameter of the method `tbox` (maybe replace with `kb`?) and also for the `Reasoner` property `self.tomasys` (maybe replace with `self.kb`?)

Rezenders, 2023-03-29:
This seems to be a feature of owlready2, not a bug. Check: https://owlready2.readthedocs.io/en/v0.37/onto.html#loading-an-ontology-from-owl-files

It says:

> The .load() method loads the ontology from a local copy or from Internet. It is safe to call .load() several times on the same ontology. It will be loaded only once.

Rezenders, 2023-09-20:
I will close this Issue since this is not a bug

closed 2023-09-20 by Rezenders