meta-control/mc_mros_reasoner #163: Allow selection of best function design to be overwritten

Rezenders, 2023-01-06:
Right now, to customize the selection of the best function design, it is necessary to modify the `obtain_best_function_design`  or `filter_fds` functions in `tomasys.py`. This is not ideal because we end up adding application specific code. My suggestion is to move these functions to the `Reasoner` class, then to customize it a new class inheriting from `RosReasoner` can be created in the application specific repos and these methods can be overridden to perform a custom selection. The new class should then be used as the mros node instead of `RosReasoner`

chcorbato, 2023-01-12:
This is an excellent proposal. It implies a bigger conceptual change in the usage of the mros software, by defining abstract classes that the library provides that need to be customized for the application. In a sense, it means giving up on putting all application-specific logic in the owl model and moving part of it into the Python code, but at least this way provides a cleaner way to do so.

I added the label `ARCH`, but maybe this is not right and we need a new label for issues related to the USAGE? of the mros library?

@Rezenders I am assigning some issues to you for the moment just for you to answer my comments. Feel free to remove yourself after doing so.

Rezenders, 2023-01-12:
I did something "similar" for the plan step of the pipeline inspection use case. Check: https://github.com/kas-lab/pipeline_inspection/blob/dev/pipeline_inspection_metacontrol/pipeline_inspection_metacontrol/pipeline_inspection_reasoner.py

> In a sense, it means giving up on putting all application-specific logic in the owl model and moving part of it into the Python code, but at least this way provides a cleaner way to do so.

Yeah

> I added the label ARCH, but maybe this is not right and we need a new label for issues related to the USAGE? of the mros library?

I think arch is fine

chcorbato, 2023-01-12:
> I did something "similar" for the plan step of the pipeline inspection use case. Check: https://github.com/kas-lab/pipeline_inspection/blob/dev/pipeline_inspection_metacontrol/pipeline_inspection_metacontrol/pipeline_inspection_reasoner.py

Nice!

This inheritance principle for the usage of the MROs library should be discussed and documented explicitly.
A point for our next MROS architecture meeting? Can you take care of it?

Rezenders, 2023-01-12:
> A point for our next MROS architecture meeting? Can you take care of it?

Yes