meta-control/mc_mros_reasoner #164: Tomasys wrapper class

Rezenders, 2023-01-07:
I suggest moving everything that directly interacts with the ontology to a TOMASysWrapper class. This would make it less confusing whether to use (tomasys or onto)/(tbox or abox), and it would be easier to replace the ontology used.

chcorbato, 2023-01-12:
Not sure I follow the rationale for the TOMASysWrapper class to make it less confusing to use tomasys or onto
With `onto` and `tomasys` I guess you mean prefixes used in the python code using owlready functions to retrieve elements from the ontology? COuld you point to a piece of code to better understand the issue?

Side note: A KB structure decision in metacontrol is to separate the tbox and the abox in different files (`tomasys.owl` and `application_name.owl`)

Rezenders, 2023-01-12:
> Not sure I follow the rationale for the TOMASysWrapper class to make it less confusing to use tomasys or onto

It is related to issue #162. We were investigating the code, and it seems that the variables are misused in some places of the code, and we are not sure why it is working. And is always a bit confusing which variable to use. So if all this information is contained in a separate class, it won't be necessary to know this if you just want to tweak the reasoner.

Another "problem" is that some code interacting directly with the ontology is at the Reasoner class. I think this should be moved to a new TOMASysWrapper class

Having a class would allow to overload its methods allowing a less ad-hoc customization

chcorbato, 2023-01-12:
> It is related to issue #162. We were investigating the code, and it seems that the variables are misused in some places of the code, and we are not sure why it is working. And is always a bit [...]

I think my reply to #162 may help explain why it is working. In the end in most cases all this parameters refer to a Python object that contains the runtime KB (ABOx and TBox included, meaning both ontology files)

> Another "problem" is that some code interacting directly with the ontology is at the Reasoner class. I think this should be moved to a new TOMASysWrapper class
> 
> Having a class would allow to overload its methods allowing a less ad-hoc customization. As mentioned in issue #163

This approach using Python classes and overloading its methods sounds good, but we need check its implications through the entire code and take the explicit decision to refactor the code around that principle systematically.
Related to #163 as you mention

Rezenders, 2024-01-17:
By the way, some time ago, I started working on this in the kb_interface branch, but I didn't finish it.
https://github.com/meta-control/mc_mros_reasoner/blob/kb_interface/mros2_reasoner/mros2_reasoner/owlready2_tomasys.py